jneilliii/OctoPrint-Domoticz #1: Can't connect to Domoticz

sepodele, 2018-07-07:
I'm getting ERROR: Invalid ip or unknown error connecting to 10.0.0.10:8090. 

[2018-07-07 16:58:37,173] ERROR: Invalid ip or unknown error connecting to 10.0.0.10:8090.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_domoticz/__init__.py", line 133, in check_status
    chk = response["result"]["Status"]
TypeError: list indices must be integers, not str
[2018-07-07 16:58:37,176] DEBUG: 10.0.0.10:8090 index 430 is UNKNOWN
[2018-07-07 16:58:37,178] DEBUG: unknown error with 10.0.0.10:8090.

jneilliii, 2018-07-07:
Can you enable the debug logging in settings and try to click the icon again, then copy/paste your plugin_domoticz_debug.log file here.

sepodele, 2018-07-07:
https://pastebin.com/ibFfmypE

jneilliii, 2018-07-07:
Ok, figured it out I think.  I just pushed an update to the repo.  Reinstall using the same url as before and once updated make sure OctoPrint restarts, or manually do it yourself and it should get past that error.

sepodele, 2018-07-07:
I use various ones but the one I'm trying to use with my 3D printer is this one: https://telldus.com/produkt/remote-controlled-outlet-telldus-z-wave/

sepodele, 2018-07-07:
Works like a charm! Thanks a bunch.

jneilliii, 2018-07-07:
Awesome.  So in order to use the M80/M81 commands or the ShutDown plugin you were commenting on before you would use M81 10.0.0.10:8090 430 in terminal or just the last part 10.0.0.10:8090 430 in the shutdown plugin and that should power off the plug.

jneilliii, 2018-07-07:
You have to make sure to also enable the GCODE options in the settings.

closed 2018-07-07 by jneilliii

jneilliii, 2018-07-08:
Would you mind testing a new version for me that integrates authentication? I'm assuming since yours worked without this you haven't set-up authentication on your Domoticz server but I want to insure that this update doesn't break it for you. Install using the following url and let me know if you don't mind. Verify that your switch is configured with a blank username and password in the plugin settings.

https://github.com/jneilliii/OctoPrint-Domoticz/archive/0.0.2.zip

jneilliii, 2018-07-08:
No need to test.  It was actually easier than I thought setting up a Dominicz server to interact with my Tasmota plug since I already had it connected to MQTT.  I'm going to be pushing an update shortly that enables authentication. I may consider taking the ip:port, username, and password out to the global settings as it's unlikely that you would be connecting to multiple servers.  What are your thoughts on that?