jneilliii/OctoPrint-Domoticz #4: Configuration

thebeetleuk, 2018-10-03:
Hi, 

I have just installed the plugin and it appears to be connecting to domoticz correctly when looking at the log. My question is how to get it to throw the switch I have? How do I set it up to turn a switch on and off?

jneilliii, 2018-10-04:
If you look in your Domoticz setup > devices section the index that is associated with your switch needs to be added into the plugin settings.  So for example from Domoticz...

[image]

Idx gets puts into the Index field of the plug configuration...

[image]

thebeetleuk, 2018-10-04:
Did all that and setup just as you have above but it doesnt seem to throw the switch. I checked in the log (today it doesnt seem to have a log file even though I just started a print - I would expect to see the call to turn on a switch reported in the log?). I'm assuming you dont need any of the boxes checked  in the piucture you have for it to work?

jneilliii, 2018-10-04:
It only creates a log file if you enable debug logging. With it the way I have in the photo above it would only trigger on clicking the button for the switch on the navbar.

thebeetleuk, 2018-10-04:
Ahhh!!! I think I follow. I didn't realise what the little lightning icon was. Ok so to get it working with GCode I need to use the instructions below...

Question tho. Why do you have to specify the IP and port here as the application wil already know what these are from the config in the picture above. Cant it just pick up the info from that instead of having to edit the Gcode? Also why the use of the M80 command? This wouldnt be a default entry in any GCODE. Would it be better to use something else (not that I'm an expert by!) I'm just thinking the plugin should run based on any GCODE instead of a manual copnfiguration on that side.

GCODE
When checked this will enable the processing of M80 and M81 commands from gcode to power on/off plug. Syntax for gcode command is M80/M81 followed by hostname/ip and index. For example if your plug is 192.168.1.2:8080 and index of 1 your gcode command would be M80 192.168.1.2:8080 1

jneilliii, 2018-10-04:
Correct, if you enable the GCODE option then the M80/M81 commands can be used to toggle the switch on/off.  The reason for requiring the IP:PORT in the command is because this was ported over from my other Tasmota plugin where there was no unique index value for your devices and the only way to find out if gcode was enabled was looping over the list of configured devices and finding the one that matched.  The reason for M80/M81 usage is because at the time of original development of the TPLink_Smartplug plugin was to mimic the [ATX power on/off](https://reprap.org/wiki/G-code#M80:_ATX_Power_On) for those of us that aren't using an ATX power supply to control. If it is presenting a problem for you because you are using this to control lights and don't want to use those commands to power off your printer because you do have an ATX power supply configured the right way to power off your printer I could include an optional `@DOMOTICZON` and `@DOMOTICZOFF` command to the code for triggering. I did do that for my TPLink-Smartplug plugin but never got around to adding that to the other power control plugins I've developed.

thebeetleuk, 2018-10-05:
ok so got them talking now :) Magic!! So now on to the Gcode. I'm not really sure what the M80 and 81 commands will do on my Prusa Mk2 so will need to run a few tests. Thaks for the support!! :)

jneilliii, 2018-10-06:
I actually pushed a new branch if you want to try it that makes the GCODE config a little easier.  Install using the url below in plugin manager.  Then for your gcode use `@DOMOTICZON` and `@DOMOTICZOFF` followed by a space and the index value.  This should prevent any conflicts with your printer with the M80 and M81 commands. Let me know how it works out for you.

https://github.com/jneilliii/OctoPrint-Domoticz/archive/0.0.3.zip

thebeetleuk, 2018-10-07:
can I use @DOMOTICZON/OFF after any command in GCODE? e.g. `M104 S210 ; set extruder temp` becomes `M104 S210 ; set extruder temp @DOMOTICZON`

jneilliii, 2018-10-07:
You should be able to but you have thi make sure to include the index from domoticz.

```
M104 S210
@DOMOTICZON 1
```

thebeetleuk, 2018-10-07:
ok so I tried running these commands form the terminal "M107 @DOMOTICZON 323" and I can't see anything in the log and it doesnt seem to work. Will it work form the terminal or do I have to load a sample gcode?

jneilliii, 2018-10-08:
Hopefully those are on two separate lines? It should work on terminal. I'll have to double check as I'll be honest I didn't test it before uploading.

jneilliii, 2018-10-08:
Ok, figured out what was going on and just pushed an update that should resolve this problem.  The commands should now work as shown below.

```
@DOMOTICZON 1 ;Turn on switch with index of 1 in Domoticz and GCODE Trigger enabled in plugin settings.
@DOMOTICZOFF 1 ;Turn off switch with index of 1 in Domoticz and GCODE Trigger enabled in plugin settings.
```

thebeetleuk, 2018-10-08:
is the update in the same version 0.0.3?

jneilliii, 2018-10-08:
Yes, just reinstall with the same url.

thebeetleuk, 2018-10-09:
bril. will try again tonight assuming my current print is finished :)

jneilliii, 2018-10-22:
It will send those commands on to the firmware either way from OctoPrint or from SD card, however I don't think that the plugin will process any of it if the file is printed from SD as the commands then don't go through OctoPrint's serial interface to be grabbed. This shouldn't cause an issue with your firmware though as the command just won't be recognized.

jneilliii, 2018-11-03:
Officially released version 0.0.3.

closed 2018-11-03 by jneilliii