jneilliii/OctoPrint-Domoticz #6: [Feature request] don't turn off power if printer is printing

calania, 2018-11-02:
Hello, I have been using this plugin for a while and it is working perfectly. I just have one problem. I have added the M81 command so that it turns off the printer 300 seconds after the print is complete. This normally works well except when I want to start a new print right after the other finished. Now I need to wait for your plugin to turn off the power before starting the new print otherwise it would kill the power mid print. What I would want is an option so that the plugin only turns off the power if the printer isn't printing. Hope this can be added!

jneilliii, 2018-11-02:
I need to check, but I think if the warn while printing option is enabled the off command won't be triggered.

jneilliii, 2018-11-02:
Good point. Looks like that branch is 2 versions ahead of the releases. I'll try to push that official tonight.

calania, 2018-11-02:
Sounds good!

closed 2018-11-02 by calania