jneilliii/OctoPrint-Domoticz #7: [Feature request] Manualy cancel gcode off delay

calania, 2018-12-13:
Hello, I really like this plugin but one small problem I have is that when I am printing something and I want to change filament right after the printer is done printing it sometimes turns off because I use the gcode delayed turn off function. Because of this, I need to wait for the printer to turn off, then turn it back on again and then change the filament. What I would want is some way of cancelling a pending shutdown. I am thinking that maybe you could make the icon in the navbar yellow while a shutdown of the printer is pending and if you were to right click on the yellow icon it cancels the pending shutdown and leaves the printer on. This is just an idea if you were to implement anything similar you could, of course, do it however you want :)

jneilliii, 2018-12-13:
Thanks for the idea.  I like the concept of the yellow -> red.  Next time I have time to look I'll see about making those changes.

calania, 2018-12-14:
Sounds good!