jneilliii/OctoPrint-Domoticz #8: Use this plugin togeter with Temperature Failsafe Plugin

calania, 2019-01-10:
Hello, I recently saw that you had made a comment on your tp link smart plug plugin in where you wrote how to make it so that the printer shuts down if it exceeds a set temp limit in case a mosfet would, for example, break https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/59 . I just wonder if it is possible to do this with the domoticz plugin and if so what should I write in the sh file?

jneilliii, 2019-01-10:
It should work the same.  Your sh file would have the following:
```
#!/bin/bash 
curl -H "X-Api-Key: <put user api key here>"  -H 'Content-Type: application/json' --data '{"command":"turnOff","ip":"<put server ip here>","idx":"<put index number here>","username":"<domoticz username>","password":"<domoticz password>"}' -X POST http://localhost:5000/api/plugin/domoticz
```

calania, 2019-01-11:
Thanks, that worked prefectly!

closed 2019-01-11 by calania