jneilliii/OctoPrint-Domoticz #9: unable to use switch to ON/OFF the printer

cyberfilou, 2019-02-06:
the plugin works half, it displays the state of the switch, but can not turn on or off the printer( with navbar switch ).
What's the problem.
here is the debug

`[2019-02-06 00:58:04,123] ERROR: Invalid ip or unknown error connecting to 192.168.1.85:8080.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_domoticz/__init__.py", line 97, in turn_on
    webresponse = urllib2.urlopen(strURL).read()
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 435, in open
    response = meth(req, response)
  File "/usr/lib/python2.7/urllib2.py", line 548, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python2.7/urllib2.py", line 473, in error
    return self._call_chain(*args)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 556, in http_error_default
    raise HTTPError(req.get_full_url(), code, msg, hdrs, fp)
HTTPError: HTTP Error 400: Bad Request
`

jneilliii, 2019-02-09:
Could you please enable debug logging in settings and then attempt a connection and then upload your plugin_domoticz_debug.log file here?

cyberfilou, 2019-02-10:
> [2019-02-10 09:46:01,619] DEBUG: {u'username': u'', u'ip': u'192.168.1.85:8080', u'password': u'', u'command': u'turnOff', u'idx': u'202 '}
[2019-02-10 09:46:01,622] DEBUG: Turning off 192.168.1.85:8080 index 202 .
[2019-02-10 09:46:01,629] ERROR: Invalid ip or unknown error connecting to 192.168.1.85:8080.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_domoticz/__init__.py", line 117, in turn_off
    webresponse = urllib2.urlopen(strURL).read()
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 435, in open
    response = meth(req, response)
  File "/usr/lib/python2.7/urllib2.py", line 548, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python2.7/urllib2.py", line 473, in error
    return self._call_chain(*args)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 556, in http_error_default
    raise HTTPError(req.get_full_url(), code, msg, hdrs, fp)
HTTPError: HTTP Error 400: Bad Request
[2019-02-10 09:46:01,631] DEBUG: Response: Unknown error turning off 192.168.1.85:8080 index 202 .
[2019-02-10 09:46:01,631] DEBUG: Unknown error turning off 192.168.1.85:8080 index 202 .
[2019-02-10 09:46:06,808] DEBUG: {u'username': u'', u'ip': u'192.168.1.85:8080', u'password': u'', u'command': u'checkStatus', u'idx': u'202 '}
[2019-02-10 09:46:06,819] DEBUG: Checking status of 192.168.1.85:8080 index 202 .
[2019-02-10 09:46:06,841] DEBUG: 192.168.1.85:8080 index 202  response: {
   "ActTime" : 1549788366,
   "AstrTwilightEnd" : "19:36",
   "AstrTwilightStart" : "06:08",
   "CivTwilightEnd" : "18:19",
   "CivTwilightStart" : "07:24",
   "DayLength" : "09:46",
   "NautTwilightEnd" : "18:58",
   "NautTwilightStart" : "06:46",
   "ServerTime" : "2019-02-10 09:46:06",
   "SunAtSouth" : "12:05",
   "Sunrise" : "07:59",
   "Sunset" : "17:45",
   "app_version" : "4.9700",
   "result" : [
      {
         "AddjMulti" : 1.0,
         "AddjMulti2" : 1.0,
         "AddjValue" : 0.0,
         "AddjValue2" : 0.0,
         "BatteryLevel" : 255,
         "CustomImage" : 0,
         "Data" : "On",
         "Description" : "",
         "DimmerType" : "none",
         "Favorite" : 0,
         "HardwareID" : 28,
         "HardwareName" : "veralite",
         "HardwareType" : "Dummy (Does nothing, use for virtual switches only)",
         "HardwareTypeVal" : 15,
         "HaveDimmer" : true,
         "HaveGroupCmd" : true,
         "HaveTimeout" : false,
         "ID" : "0001411A",
         "Image" : "Light",
         "IsSubDevice" : false,
         "LastUpdate" : "2019-02-07 18:40:50",
         "Level" : 0,
         "LevelInt" : 0,
         "MaxDimLevel" : 100,
         "Name" : "Imprimante 3d",
         "Notifications" : "false",
         "PlanID" : "0",
         "PlanIDs" : [ 0 ],
         "Protected" : false,
         "ShowNotifications" : true,
         "SignalLevel" : "-",
         "Status" : "On",
         "StrParam1" : "aHR0cDovLzE5Mi4xNjguMS44MTozNDgwL2RhdGFfcmVxdWVzdD9pZD1hY3Rpb24mb3V0cHV0X2Zvcm1hdD14bWwmRGV2aWNlTnVtPTI3OSZzZXJ2aWNlSWQ9dXJuOnVwbnAtb3JnOnNlcnZpY2VJZDpTd2l0Y2hQb3dlcjEmYWN0aW9uPVNldFRhcmdldCZuZXdUYXJnZXRWYWx1ZT0x",
         "StrParam2" : "aHR0cDovLzE5Mi4xNjguMS44MTozNDgwL2RhdGFfcmVxdWVzdD9pZD1hY3Rpb24mb3V0cHV0X2Zvcm1hdD14bWwmRGV2aWNlTnVtPTI3OSZzZXJ2aWNlSWQ9dXJuOnVwbnAtb3JnOnNlcnZpY2VJZDpTd2l0Y2hQb3dlcjEmYWN0aW9uPVNldFRhcmdldCZuZXdUYXJnZXRWYWx1ZT0w",
         "SubType" : "Switch",
         "SwitchType" : "On/Off",
         "SwitchTypeVal" : 0,
         "Timers" : "false",
         "Type" : "Light/Switch",
         "TypeImg" : "lightbulb",
         "Unit" : 1,
         "Used" : 1,
         "UsedByCamera" : false,
         "XOffset" : "0",
         "YOffset" : "0",
         "idx" : "202"
      }
   ],
   "status" : "OK",
   "title" : "Devices"
}

[2019-02-10 09:46:06,843] DEBUG: 192.168.1.85:8080 index 202  is On
[2019-02-10 09:46:11,738] DEBUG: {u'username': u'', u'ip': u'192.168.1.85:8080', u'password': u'', u'command': u'turnOff', u'idx': u'202 '}
[2019-02-10 09:46:11,739] DEBUG: Turning off 192.168.1.85:8080 index 202 .
[2019-02-10 09:46:11,745] ERROR: Invalid ip or unknown error connecting to 192.168.1.85:8080.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_domoticz/__init__.py", line 117, in turn_off
    webresponse = urllib2.urlopen(strURL).read()
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 435, in open
    response = meth(req, response)
  File "/usr/lib/python2.7/urllib2.py", line 548, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python2.7/urllib2.py", line 473, in error
    return self._call_chain(*args)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 556, in http_error_default
    raise HTTPError(req.get_full_url(), code, msg, hdrs, fp)
HTTPError: HTTP Error 400: Bad Request
[2019-02-10 09:46:11,747] DEBUG: Response: Unknown error turning off 192.168.1.85:8080 index 202 .
[2019-02-10 09:46:11,748] DEBUG: Unknown error turning off 192.168.1.85:8080 index 202 .

>

closed 2019-03-03 by stale[bot]