jneilliii/OctoPrint-Domoticz #11: ERROR: Invalid ip or unknown error connecting to 192.168.1.248:8080.

chatainsim, 2019-10-22:
Hello,
Since the last update of the plugin, it failed to connect to Domoticz, here is the log:
```
[2019-10-22 22:01:04,268] ERROR: Invalid ip or unknown error connecting to 192.168.1.248:8080.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_domoticz/__init__.py", line 148, in check_status
    response = json.loads(webresponse)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
TypeError: expected string or buffer
```

chatainsim, 2019-10-22:
Debug logs:
```
[2019-10-22 22:07:03,398] DEBUG: 192.168.1.248:8080 index 381 is UNKNOWN
[2019-10-22 22:07:03,398] DEBUG: unknown error with 192.168.1.248:8080.
[2019-10-22 22:07:28,591] DEBUG: {u'username': u'username', u'ip': u'192.168.1.248:8080', u'password': u'password', u'command': u'checkStatus', u'idx': u'381'}
[2019-10-22 22:07:28,592] DEBUG: Using authentication for 192.168.1.248:8080.
[2019-10-22 22:07:28,593] DEBUG: Checking status of 192.168.1.248:8080 index 381.
[2019-10-22 22:07:28,612] DEBUG: 192.168.1.248:8080 index 381 response: <Response [200]>
[2019-10-22 22:07:28,613] ERROR: Invalid ip or unknown error connecting to 192.168.1.248:8080.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_domoticz/__init__.py", line 148, in check_status
    response = json.loads(webresponse)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
TypeError: expected string or buffer
[2019-10-22 22:07:28,613] DEBUG: 192.168.1.248:8080 index 381 is UNKNOWN
[2019-10-22 22:07:28,614] DEBUG: unknown error with 192.168.1.248:8080.
```

jneilliii, 2019-10-22:
Thanks I'll take a look into it.

jneilliii, 2019-10-22:
I just released version 0.1.1 that should fix this error.

closed 2019-10-22 by jneilliii

chatainsim, 2019-10-23:
Thanks, it's fixed now.