jneilliii/OctoPrint-Domoticz #14: Auto Connect not working via API (Cura Otcoprint Plugin)

Wob76, 2020-03-26:
Running Cura 4.5 and the Connect to OctoPrint 3.5.12, Octoprint 1.4.0, OctoPrint-Domoticz 0.1.1

This combination is working to power the printer up, but it doesn't initiate the auto connect after the power up. I have the Domoticz plugin configured to "Auto Connect" and GCODE on Delay, I have also tried adjusting the delays without change. The plugin does work as expected when using the Octoprint plugin Power Icon, or inserted start GCODE.

I opened an issue in the Cura Plugin and author reported that it appear this plugin does not issue a AutoConnect when using the API, could this be looked at? 
https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/164#issuecomment-604291797

Thanks,
Beau

jneilliii, 2020-03-26:
@fieldsOfView is correct. The current implementation of the plugin handles the connection on the JavaScript/Web Interface side and not the python/server side. I'll need to make these adjustments the way I have in my other power plugins to resolve this.

Wob76, 2020-03-27:
That would be great, it's not a major issue, I just have to open otcoprint and connect, but the updates with Cura are making the integration much nicer, this would mean I need very little interaction within octoprint to start a print.

jneilliii, 2020-03-27:
Hey @Wob76, I just pushed an update to the 0.1.2 branch for testing. Use the URL below in Plugin Manager to install and verify for me prior to official release. After install please make sure to force a refresh of the browser is not prompted to Reload after installing/restarting.

https://github.com/jneilliii/OctoPrint-Domoticz/archive/0.1.2.zip

Wob76, 2020-03-27:
Great, I'll give it a spin

Wob76, 2020-03-27:
Works a treat, powered up and auto connected as expected, haven't seen any issues.

PR #15 (2020-03-27): 0.1.2
Move system commands and connect/disconnect logic to server side processing, resolves #14
closed 2020-03-27 by jneilliii