jneilliii/OctoPrint-Domoticz #16: Scenes

chatainsim, 2020-04-07:
Hello,
Thanks for this great plugin.
Is it possible to switch on/off a groupe defined in scenes?
I've want to turn on and off my Ender 3 and my led at the same time.
Thanks

jneilliii, 2020-04-07:
I've never tried, assuming the scene has an index I would suspect it would work. I personally don't use this plugin or domoticz.

chatainsim, 2020-04-07:
Ok, strange to make a plugin but don't use it.
API is here: https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s#Turn_a_scene_.2F_group_on_or_off

```
/json.htm?type=command&param=switchscene&idx=&switchcmd=

    idx = id of your scene/group.
    switchcmd = "On", "Off" or "Toggle" (case sensitive!)
    Scenes can only be turned On
```
I will have a look later and try to make a PR.

jneilliii, 2020-04-07:
I agree, it was a little strange but the plugin was created by request from someone that saw my TPLink plugin. It wasn't too difficult to achieve. Based on your comment above though, if a scene can only be turned on, what's the point? Don't know if that's a documentation issue on their site or not since they say swtchcmd can be on, off, or toggle.

chatainsim, 2020-04-08:
Because there are two type of Scenes, group and scene.
Yes I know it's wired. Scene can only be turn on but a group can be turn on and off.
Thanks

jneilliii, 2020-04-08:
Cool. Happy to accept a pull request, the changes would probably be minimal.

countcobolt, 2021-04-10:
I just checked a bit of the code, but to be honest, my Python is c.r.A.p. So you seem to use strUrl with switchLight. For setting a scene it would be /json.htm?type=command&param=switchscene&idx=**8**&switchcmd=**On** (or "Off" in case you wanted off) The reason I didn't fiddle with it, is that I believe you would need to add option in the interface to select if it is a switch of a scene. The main reason I use a scene, is that I have 2 sonoff switches (1 for Rpi, 1 for printer) and the scene itself turns of the rpi at 1 minute, while the printer after 10 minutes.

cyberfilou, 2021-11-22:
@countcobolt why don't you try using Block Events, it should be pretty easy to do. If switch ON then after X sec light on ......