jneilliii/OctoPrint-Domoticz #17: Invalid ip or unknown error connecting to 192....

kylana1, 2020-04-21:
Hello

my domoticz server not detected

[2020-04-21 22:34:41,377] ERROR: Invalid ip or unknown error connecting to 192.168.1.32:8181.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_domoticz/__init__.py", line 165, in check_status
    chk = response["result"][0]["Status"]
KeyError: 'result'

I try with full new installation and with local domoticz (127.0.0.1) so it's same issue

jneilliii, 2020-04-22:
If you're getting a key error there in that bit of code then you are not getting an understandable response from the server. That could be that your id is incorrect possibly or you haven't include the login parameters maybe?

kylana1, 2020-04-22:
hello thanks for quick answer.
I don't use password on local network.
but, I tested with new user create for octoprint

jneilliii, 2020-04-22:
I think that might be the issue on your domoticz side the device needs to be of type "Light/Switch" to function properly.

kylana1, 2020-04-26:
Thanks it's done

closed 2020-04-26 by jneilliii