jneilliii/OctoPrint-Domoticz #18: Protected (passcode) switches are not supported

SwiftingSpeed, 2020-07-10:
I have my 3D-printer switch set as protected in Domoticz to prevent it being accidentally powered off in Domoticz UI, this means that an additional passcode needs to be supplied for the switch to turn off or on. Unfortunately This plugin does not support the protected function as a passcode can't be entered or supplied with the Domoticz json command.

The parameter is called "passcode" and is passed in the url just like all other parameters so it should not be that hard to add it.
Here's Domoticz reference: https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s#Turn_a_light.2Fswitch_on

> /json.htm?type=command&param=switchlight&idx=99&switchcmd=Off&passcode=YOUR_SWITCH_PASSWORD_AS_SET_IN_SETTINGS

I have manually edited the `__init__.py` file to add the passcode and confirmed that the plugin works, both getting the switch status (red or green icon) and can be switched on and off. For example line 94:
https://github.com/jneilliii/OctoPrint-Domoticz/blob/db3e92952362a85f24d34082b94b1cb9c9685aeb/octoprint_domoticz/__init__.py#L94
to
`strURL = "http://" + plugip + "/json.htm?type=command&param=switchlight&idx=" + str(plugidx) + "&switchcmd=On&passcode=1234"`
This should preferably be a configurable passcode.

jneilliii, 2020-07-10:
Thanks for the fix, I'll integrate that as a setting.

jneilliii, 2021-04-10:
@SwiftingSpeed I've added the passcode options with the above commit. If you are still using this plugin and would like to try it out you can copy/paste the URL below into plugin manager > get more > ...from URL and clicking install over your existing install.

```
https://github.com/jneilliii/OctoPrint-Domoticz/archive/refs/tags/0.1.4rc1.zip
```

SwiftingSpeed, 2021-04-10:
Great! I'll see if I can try it soon.

NiklasOlofsson, 2022-06-26:
Working as expected with protected switches. Btw, ID could be replaced as IDX. In Domoticz you work with the IDX of devices. The ID-field is used for other stuff and could contain letters and special chars. IDX is always a unique number.

Please, if you could, add a "Save" button when editing a switch in the UI (i'm using UI Customizer). If I change a switch and press close, it often does not save and I need to go back several times for my changes to stick.

jneilliii, 2022-06-26:
Don't press close, after editing a device use the save button in main settings window.

NiklasOlofsson, 2022-06-26:
> Don't press close, after editing a device use the save button in main settings window.

The button in the background is inactive when the window is open. Clinking on save (or anywhere outside the box) will close the switch i'm editing and then I can click Save for real. Don't know if it's an UI Customizer thing.

jneilliii, 2022-06-26:
no, that is the expected behavior. the close button on the device edit window will close, but if you don't press the save button in the main settings window it will revert the settings. In my other plugins I've even added a note to the edit window to indicate that like below. 

[image]

NiklasOlofsson, 2022-06-26:
> no, that is the expected behavior.

Ah. I must have left the settings without Saving. Thanks!

closed 2025-06-29 by jneilliii