jneilliii/OctoPrint-Domoticz #19: Connecting

mattx38, 2020-09-08:
Hello,

Cannot manage to connect OctoPrint with my Domoticz server.

My Domoticz private IP is http://192.168.0.100:8080, but it keep saying me that Domoticz status is unknown and thaht I have to double check its IP/hostname.

I have no password to connect to Domoticz.
What should I check ?

yellowpanther, 2020-09-12:
Hello @mattx38,

Are you using python2 or python3 ?

Check the log file : ~/.octoprint/logs/plugin_domoticz_debug.log
If there is nothing in the file, check "Enable debug logging." in octoprint's domoticz plugin settings, restart octoprint and check the log file.

jneilliii, 2020-09-12:
Thanks @yellowpanther for the assist. You may have to restart OctoPrint after enabling the debug logging for everything to initialize properly.

mattx38, 2020-09-14:
Hello, 

My Python version is 2.7.16

Matt

mattx38, 2020-09-14:
I did add the following line in confit.js :
debugMode: true

But no logs in MagicMirror/modules/octomirror-module at all

Matt

yellowpanther, 2020-09-14:
I think you got the wrong thread ;)
This is the plugin for domoticz.

mattx38, 2020-09-14:
Hello,

My bad !
You can close this issue. 

Matt

mattx38, 2020-09-14:
Sorry and thank you for all you did. 

Matt

closed 2020-09-14 by jneilliii