jneilliii/OctoPrint-Domoticz #20: Plugin failed since updated Octoprint to python3

chatainsim, 2020-10-14:
Hello,
I've upgraded my octoprint to use python3.
Since the Domoticz Plugin failed:
```
2020-10-14 18:52:40,644] DEBUG: {'command': 'checkStatus', 'ip': '192.168.1.248:8080', 'idx': '441', 'username': 'simon', 'password': 'xxxx'}
[2020-10-14 18:52:40,644] DEBUG: Using authentication for 192.168.1.248:8080.
[2020-10-14 18:52:40,645] DEBUG: Checking status of 192.168.1.248:8080 index 441.
[2020-10-14 18:52:40,645] ERROR: Invalid ip or unknown error connecting to 192.168.1.248:8080.
Traceback (most recent call last):
  File "/home/simon/OctoPy3/lib/python3.7/site-packages/octoprint_domoticz/__init__.py", line 161, in check_status
    strURL = strURL + "&username=" + base64.b64encode(bytes(username)) + "&password=" + base64.b64encode(bytes(password))
TypeError: string argument without an encoding
[2020-10-14 18:52:40,645] DEBUG: 192.168.1.248:8080 index 441 is UNKNOWN
[2020-10-14 18:52:40,645] DEBUG: unknown error with 192.168.1.248:8080.
```
Can you help me with this ?
Thank you

jneilliii, 2020-10-14:
I'll have to dig into that error. in python 3 to see what it's doing, but will get to it eventually.

chatainsim, 2020-10-15:
Ok, I've tried to install Octoprint with Python on another Linux and there is no such error.
I think there is something wrong with my migrated Octoprint.
I'll dig into it and get back to you.

Thank you.

chatainsim, 2020-10-16:
I haven't the issue on new install because login/pass was left empty.
Maybe something like that can fix the issue:
```
base64.b64encode(bytes(username, encoding='utf8'))
```
or
```
base64.b64encode(bytes(username, 'ascii')
```

jneilliii, 2020-10-16:
Thanks for the legwork on this. The issue is definitely related to that line, but I need to make sure that the change doesn't effect the python2 implementation as well. I'll try to do some testing over the weekend to get this resolved.

jneilliii, 2020-10-16:
I wonder if you could help remind me where that password is configured in domoticz? I don't use it regularly and can't seem to find it.

jneilliii, 2020-10-16:
I found it and just released a new version that should resolve this issue for you.

commit ad323322 (2020-10-16): fix python3 related issue with username password encoding, resolves #20
closed 2020-10-16 by jneilliii