jneilliii/OctoPrint-Domoticz #21: Domoticz Error

varazir, 2020-11-17:
Hello 

I'm getting this error: 
Domoticz Error
Status unknown for 192.168.0.10:80. Double check IP Address\Hostname in Domoticz Settings.

Log files 
[plugin_domoticz_debug_20201117.log](https://github.com/jneilliii/OctoPrint-Domoticz/files/5555719/plugin_domoticz_debug_20201117.log)
[octoprint_20201117.log](https://github.com/jneilliii/OctoPrint-Domoticz/files/5555720/octoprint_20201117.log)

Domoticz (0.1.3)
 Homepage  jneilliii
Simple plugin to work with Domoticz to control lights.

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.198 Safari/537.36
connectivity.connection_check: 192.168.20.20:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 915709952
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x80000
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.0rc2

jneilliii, 2020-11-17:
Your domoticz version seems quite old. I've tested this plugin with domoticz Version 4.10717 (May 9th 2019) just now here and everything seems to be working fine in the Octoprint 1.5.0rc1 and OctoPrint 1.5.0rc2, under python 3. I see you're running python 2 so I was setting the plugin up on another pi for testing, but initially looking at your log it appears that the server isn't accepting the connection. You sure you have the right ip:port configured? The default for domoticz is 8080 isn't it?

varazir, 2020-11-17:
> Your domoticz version seems quite old. I've tested this plugin with domoticz Version 4.10717 (May 9th 2019) just now here and everything seems to be working fine in the Octoprint 1.5.0rc1 and [...]

Hmm I'm running beta on my Domoticz 
Version: 2020.2 (build 12644)
Build Hash: e3044d2e1-modified
Compile Date: 2020-11-15 10:04:26
dzVents Version: 3.0.16
Python Version: 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]

I can ping the Ip from the command line on my octoprint . 

I'm running my domoticz on port 80

jneilliii, 2020-11-18:
> Hmm I'm running beta on my Domoticz

Ok, I'll have to look into this further with that newer version. I tried to install the newer stable domoticz update but my virus protection flagged it. Did this plugin work for you in the previous stable octoprint version?

varazir, 2020-11-18:
> > Hmm I'm running beta on my Domoticz
> 
> Ok, I'll have to look into this further with that newer version. I tried to install the newer stable domoticz update but my virus protection flagged it. Did this plugin work for you in the previous stable octoprint version?

I can only find logs after RC1 I think 

[plugin_domoticz_debug.log.zip](https://github.com/jneilliii/OctoPrint-Domoticz/files/5559158/plugin_domoticz_debug.log.zip)

varazir, 2020-11-18:
The some setting in the plugin kept crashing my octoprint. Cleared the data and now it's working

varazir, 2020-11-18:
> > Hmm I'm running beta on my Domoticz
> 
> Ok, I'll have to look into this further with that newer version. I tried to install the newer stable domoticz update but my virus protection flagged it. Did this plugin work for you in the previous stable octoprint version?

I think so.

jneilliii, 2020-11-19:
Ok, but now the plugin is working as expected?

varazir, 2020-11-19:
> Ok, but now the plugin is working as expected?

When I add my IP it starts to crash my octoprint.

jneilliii, 2020-11-19:
Ok, thanks. I'll have to re-create your setup with the beta domoticz version then on port 80, because I am not seeing the same thing here.

jneilliii, 2020-11-19:
Quick question. Have you tried without the port specified and just used the ip? Since you're running on port 80 I don't think it's necessary to include the port number, although that shouldn't block it from working.

varazir, 2020-11-19:
> Quick question. Have you tried without the port specified and just used the ip? Since you're running on port 80 I don't think it's necessary to include the port number, although that shouldn't block it from working.

Worked without portnumber

jneilliii, 2020-11-19:
Yay, so that must be something with how the connection is made when using standard web port. If all is good now, I think I can close this issue? I will add a note to the readme that explains this.

varazir, 2020-11-19:
Sure

closed 2020-11-19 by jneilliii