jneilliii/OctoPrint-Domoticz #24: Plugin not working with Domoticz 2023.1

J0uj0u, 2023-03-03:
New security rule for API use : 
API URLs with username and password as parameters (eg https://IP:PORT/json.htm?username=userBASE64&password=PWDBASE64& ) are not supported anymore!

Use https://username:password@IP:PORT/json.htm instead.

Can you update it ?

jneilliii, 2023-03-03:
Thanks for supplying the required changes. I can make that as an option in a future update. Shouldn't take too long to do, and I might be able to make those changes tonight.

jneilliii, 2023-03-04:
Implemented in the above commit and made available as version 0.1.4rc2. switch the release channel for the Plugin to `Release Candidate` in OctoPrint's Software Update settings and upgrade when prompted. You will need to enable this option.

[image]

J0uj0u, 2023-03-04:
Great ! Thanks for the quick fix

It work well

adrienafl, 2023-06-01:
> Implemented in the above commit and made available as version 0.1.4rc2. switch the release channel for the plugin to `Release Candidate` in OctoPrint's Software Update settings and upgrade when prompted. You will need to enable this option.
> 
> [image]

Yay, just found this issue I was trying to solve for an hour. However, I don't see the option to change the release chanel for domoticz plugin (I do see it for others plugins). Am I missing something ?

adrienafl, 2023-06-01:
Kept investigating. My setup as a FYI : Ubuntu server running everything on docker

After whitelisting docker IP range on domoticz, the plugin worked again. The issue must be somewhere in the new authentication process. It's a bit late now to try to read and debug your code.

I'm also using homebridge and I managed to have domoticz connection working with auth. That's why I'm quite sure it's an auth issue on your plugin.

Million thanks for the time you'll spend on it :)

jneilliii, 2023-06-01:
> After whitelisting docker IP range on domoticz, the plugin worked again. The issue must be somewhere in the new authentication process. It's a bit late now to try to read and debug your code.

If this resolved the issue than I suspect it was something that changed in how domoticz was handling "local access" and not how the plugin was authenticating.

adrienafl, 2023-06-01:
I don't follow you. The allowlist I did was made from the Security panel of domoticz. This setting allows you to disable authentication for specific ranges. When it wasn't enabled, I was able to curl my domoticz IP from octoprint, this wasn't the issue.

jneilliii, 2023-06-01:
Oh, I see what you mean now. You whitelisted the IP range to not require authentication. I will have to re-review then as I don't use domoticz regularly. Will check when I have time available.

closed 2025-06-29 by jneilliii